Gulivertx/cbatmo #1: Charts screen not working

jukkaforss, 2019-01-30:
I exported to raspberry pi and got it partly working.
Screen that has time and weather, works perfectly.
But charts screen is only generating "Something went wrong" message.

I don't have all modules, only basic main module and external module.
In running log I don't get any errors.

Is this issue caused by missing modules and it there a way to remove those from configuration.
I am not expert in javascript coding and every time I try to change code, I break it.

[image]

jukkaforss, 2019-01-30:
I build it in x64 linux and copied whole directory to raspberry pi.
Then I needed to recompile node-sass,  cwebp-bin, optipng-bin and pngquant-bin modules.
Then just nom run dev:start

Gulivertx, 2019-01-30:
Hi,

First off all thank you for email.

About your question, you effectively got the error “something went wrong” because you do not have all similar modules as main.
This what I’m speaking inside the README by telling “there is not intelligence” to find existing modules….

To quick solve your problem you just have to edit this file : 
https://github.com/Gulivertx/cbatmo/blob/master/client/components/Netatmo.js
Tee goal is to remove all modules that you do not have with you station then for that remove the lines as following:
First this lines https://github.com/Gulivertx/cbatmo/blob/6ec77d98f9f38769ff19ac905367d95b63684463/client/components/Netatmo.js#L134-L218
Second this line https://github.com/Gulivertx/cbatmo/blob/6ec77d98f9f38769ff19ac905367d95b63684463/client/components/Netatmo.js#L263-L298

After that should have only declaration for modules “Main” and “External”.
Then run “npm run dev:start” to see if every thing compile correctly and if the error disappears.

Just for you knowledge, you MUST not build on Raspberry but only from your Linux x64 computer. As you see the Raspberry does not have some NPM packages as node-sass, ….
So basically in your linux station you do everything, configure your APIs keys for netatmo and darksky, install package dependancies with npm install, make the corrections of Netmo.js file as descibe upper in this email.
When you fill ready, do a try on the same computer with npm tun start:dev, reach the server with google chrome or firefox in the address http://localhost:3000, open the dev tools ans look after any error in the javascript console.
If you do not have any errors and you do not get anymore the message “something went wrong” go ahead and deploy to your raspberry.

A short way from your Linux computer is :
Run the command to build the app (this time not as dev but as production build” then for that “npm run build”, then run the script deploy.sh with this command “sh deploy.sh” (if the script do nothing, maybe you have to give it executable access with the command chmod a+x deploy.sh” and run again “sh deploy.sh”.
This script should create an archiive tar.gz in the parent folder of the project with a name as “cbatmo-1.2.0.tar.gz”. Upload this archive on your Raspberry with ssh or something else. Then log with ssh into your raspberry and unarchive this “cbatmo-1.2.0.tar.gz” file. Then go into the new created folder with “cd cbatmo” and then install package dependancies but this time not with npm install but with “npm install —production”. This command do not install missing packages on raspberry as node—sass. FInally run the command “npm start” to start the production server, GOAL, here you are !!! You should now able to open with your raspberry the URL http://localhost:3000.

I you still need any help do not hesitate to contact me back.

Kind Regards,

Cédric

jukkaforss, 2019-01-31:
There is of course possibility that I have so little data and page is dynamic so that makes width so small.

jukkaforss, 2019-01-31:
I got it fixed, I changed _netatmo.scss file

I increased chart hight, changed locations and configurations of next components.

.chart-weather-line {
  max-height: 155px;
  height: 155px;
  width: 100%;
}

  .humidity {
    position: absolute;
    top: 64px;
    font-size: 16px;
    font-weight: 400;
    color: $white;

    .wi {
      color: $blue;
    }
  }

  .noise {
    position: absolute;
    top: 108px;
    font-size: 16px;
    font-weight: 400;
    color: $white;

    .wi {
      color: $purpule;
    }
  }

.module-main {
  .col1 {
    width: 550px;
    float: left;
  }

  .col2 {
    width: 200px;
    float: left;
  }
}

.module-external {
  .col1 {
    width: 550px;
    float: left;
  }

  .col2 {
    width: 200px;
    float: left;
  }
}

[image]

Gulivertx, 2019-01-31:
Good job!
I know that I have to rework some parts of the app, as a test of existing modules before the first render and only render charts for modules detected.
I should also have to rework to use for instance flex-box for chart box positioning and stretching. With these changes the major troubles should go away...

I also know that there is a bug when netatmo api or dark sky respond with an error... If I found time I would maybe update these, let's see....

closed 2019-01-31 by Gulivertx