Gulivertx/cbatmo #2: Error lookup "index"

Yezariael75, 2019-09-27:
Hello. Many thanks for your effort! As of the screenshots that's exactly what I am looking for :) 

Now, your description is not correct in many ways and describes not the whole process. I have no clue in linux, Java or else. So it was a hard way but I could figure out how to get the app installed.

Unfortunately, when going to the site localhost:3000 for DEV or same in PROD I get an error message: 

Failed to lookup view "index" in views directory "/home/pi/cbatmo/views"

I assume that files are missing? 

Thanks for you help!
Alexander

Gulivertx, 2019-09-28:
The instruction for DEV seems to be complete. If you do not have index.ejs in the folder views it seems that you do not compile the project.

Did you run the command : `yarn run dev:start ` ?
You must keep the command running to reach the http://localhost:3000 to have a web server serving the app.

here is the quick steps to run forward the app :

- `git clone https://github.com/Gulivertx/cbatmo.git && cd cbatmo?
- `yarn`
- Edit `config/api.json`
- `yarn run dev:start` // wait until the end of the compilation, may take 15s, you console will write a lot of think when the compilation is OK (but not kill the process, keep it running)
- look in `views` folder if `index.ejs` was created correctly

Then if it is ok open you web browser and go to `http://localhost:3000`

Yezariael75, 2019-10-04:
Thanks for your answer. I tried it from a Windows Workstation, there are a lot of things different. I will install an Ubuntu VM and try it again.

Gulivertx, 2019-10-04:
Windows, MacOS or Linux is about the same business for this kind of app. 
Just install NodeJS, Yarn and git on your Windows and you are able to use this app with the commands above.

karba71, 2019-10-06:
Hello Gulivertx,

cbatmo is just the thing i'm looking for! Great!!!

I have install node js on my Windows machine and then clone cbatmo.
There are some probs with yarn and npm:

C:\Users\Karst\cbatmo>yarn
Der Befehl "yarn" ist entweder falsch geschrieben oder konnte nicht gefunden werden.

C:\Users\Karst\cbatmo>npm install
npm WARN deprecated loadable-components@2.2.3: Please use @loadable/component
.
.
.
Binary found at C:\Users\Karst\cbatmo\node_modules\node-sass\vendor\win32-x64-64\binding.node
Testing binary
Binary is fine
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules\fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"win32","arch":"x64"})

added 1130 packages from 566 contributors and audited 12264 packages in 61.978s
found 0 vulnerabilities

C:\Users\Karst\cbatmo>npm run dev:start

> cbatmo@2.0.0 dev:start C:\Users\Karst\cbatmo
> NODE_ENV=development concurrently -k "webpack --watch --mode development" "node server.js"

Der Befehl "NODE_ENV" ist entweder falsch geschrieben oder konnte nicht gefunden werden.
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! cbatmo@2.0.0 dev:start: `NODE_ENV=development concurrently -k "webpack --watch --mode development" "node server.js"`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the cbatmo@2.0.0 dev:start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\Karst\AppData\Roaming\npm-cache\_logs\2019-10-06T09_33_44_875Z-debug.log

What is wrong?

Yezariael75, 2019-10-06:
> Der Befehl "NODE_ENV" ist entweder falsch geschrieben oder konnte nicht gefunden werden.>

Exactly what I said, on windows a lot of things are different. For example windows needs "SET NODE_ENV" to get it work.

Yezariael75, 2019-10-06:
I just tried it on an Ubuntu VM, there it works just fine :) 

Don't do it with a Windows machine.

Gulivertx, 2019-10-11:
OK I did a try on my Windows Workstation and get some troubles. The problem is about WIndows do not set variable as MacOS or Linux and not reconize internal project package bynaries.

To start the dev mode replace `yarn run dev:start` by these two commands : 

- `set NODE_ENV=development`
- `node_modules\\.bin\\concurrently -k "node_modules\\.bin\\webpack --watch --mode development" "node server.js"`

closed 2019-10-17 by Yezariael75