Gulivertx/cbatmo #4: Problems with the deploy.sh step

Lukili1989, 2019-12-16:
Hy,
In the Browser the programm is running. Now i want to get it to my Raspberry but i don´t can create the file for the Raspberry. Please can you help me.

./deploy.sh how i can use this file to create the .tar.gz. archiv

This is my first Projekt with Node.js and i don´t know how to use i´m only trying and now i don´t can find why it don´t run

Best regards Lichti

Gulivertx, 2019-12-17:
hey there ! Thank you for your interest of CBatmo.

Ok, sounds good, cbatmo is running without any trouble in your main computer if I have understand ? Good point, mainly if you're new with nodejs.

To help you I'll need more information. I mean which steps did you try to deploy the app in your Raspberry pi ?
Could you tell me what is you main computer OS ?
Which is the target system OS in your PI ?

What I can quick tell you is that you need more and less doing these following steps :

- On your computer build cbatmo for production `yarn run build`
- Run the script `deploy.sh` which will create a tar.gz archive
- upload this archive to your raspberry, you can use for instance sftp (or whatever else, as an USB key)
- uncompress the archive on your PI
- Install nodejs on your pi
- run the webserver from your pi `cd cbatmo && yarn start`(this is just an example and depending of your path could not work)

Then if no error you should be able to reach the app from you main computer to your raspberry with the URL `http://PI_IP_ADRESSS:3000`

If it's still work you have to configure your Pi to run the nodejs webserver and open a new webpage on boot on the url `http://localhost:3000`. 

To run the server at boot the best way is to use systemd and create a new service to launch the nodejs webserver at Pi boot.

Then you should have a look how-to run your Pi as a kiosk mod to GET the best experience. Like using [yocto](https://www.yoctoproject.org/) project (which is my favorite way but really not easy), or using a solution like this https://medium.com/@WebReflection/a-minimalistic-64-bit-web-kiosk-for-rpi-3-98e460419b47 (this was my old way but I replaced Xorg by Wayland). or maybe more simple at first using a Raspbian opening LXDE with an autostart to Chrome which is configured to run as kiosk mod in advanced configuration and opening http://localhost:3000 by default (you will still need for all of these the webserver running at first)

Lukili1989, 2019-12-22:
Hi thanks for your help.

I have tried a lot of things and now i had run it on my raspberry!!

I have one more question.
I will run it with an autostart script because i will make this for my dad and he don´t know anything about raspberry. So if he give the raspberry power all should run after boot the system first the "npm start" should run than the browser should open.
To open browser thats not the Problem but to run the start part i don´t can handle it please can you help me.
My Computer OS is Windows.
And my Raspberry Pi OS is Raspian.

Thanks Luki

Gulivertx, 2019-12-23:
to start the webserver you should use systemd init on your Raspbian.

I created one init script as example which can be found in the branch **2.1.x**. 
https://github.com/Gulivertx/cbatmo/blob/2.1.x/rpi/cbatmo.service

Add this script in the system folder of your systemd. If Raspbian is as ArchLinux your should add this file in the following path : `/etc/systemd/system/`.

Then load this init script in systemd and each time the RPI will boot the webserver will be launching automatically during the boot process.

To load the init script : `systemctl enable cbatmo.service`
To check if the server run correctly : `systemctl status cbatmo`

**You must modify some value of the script before using it**, edit it and change these :

- WorkingDirectory=/home/alarm/cbatmo
- User=alarm
- Group=alarm

Gulivertx, 2020-02-24:
As no return from you and also a new version of deploy script is available and should help for this kind of problem I close this issue.

closed 2020-02-24 by Gulivertx