Gulivertx/cbatmo #8: No Data Under Chambre of Mathias

coldstream, 2020-01-04:
Hi!
First! I appreciate the work you put into this!  I had quite a bit of fun figuring out how to get this on my Raspberry Pi (I bought the LCD screen after I saw this).  

For the issue, I'm out of ideas on why I don't see any weather data populating in the "Chambre of Mathias(sp?)" area.  It is blank here on my setup.  All the other fields are working great.  I am guessing this data is supposed to be populated by Darksky?  

If you need further info, please let me know!

Todd Osborne

Gulivertx, 2020-01-04:
Hi, the missing data is an indoor additional Netatmo module. 
If you not have any additional indoor module this is normal to have this blank area. 
What I could do easily it’s to extend the rain module graph to use the whole available area if there is not any additional indoor module detected. 
Or if you want to propose a code correction to handle this it will also be great :)

coldstream, 2020-01-11:
Hi!
I think if an additional module is missing, perhaps move the WIND up to RAIN GAUGE.  Then stretch the GRAPH section to accommodate the empty space left.  (see my image I uploaded for reference)

I think this is the best approach because the GRAPH can be changed depending on what module you tap on (RAIN, WIND, TEMP, etc.).  Keeping it somewhat separate might be more intuitive for pp. to change the graph.  My 2 cents!

Todd

Gulivertx, 2020-06-05:
I close this issue as it related to https://github.com/Gulivertx/cbatmo/issues/13 which will also fix this.

closed 2020-06-05 by Gulivertx