Gulivertx/cbatmo #9: Inconsistencies in Unit Type (F vs C)

coldstream, 2020-01-04:
Opening a separate issue.  Some of my data is in Celsius while my forecast units are in Fahrenheit. Also, my Wind is in Imperial units (which is what I want). The current Indoor/Outdoor temperature readings are in Celsius.  I prefer to have them in Fahrenheit.  Please advise if I need to do something on my end to make them all in Fahrenheit.

Gulivertx, 2020-01-04:
I need to investigate this, it seems that I forget to handle temperature unit... I will check this the next week

Gulivertx, 2020-02-09:
Currently looking for this issue and it seems that Netatmo API do not return the values according to the user units settings.
I asked on the Netatmo forum, will see if I get an answer, but I feel like I will have to convert values myself.... Let's see.
https://forum.netatmo.com/viewtopic.php?f=5&t=17892

Gulivertx, 2020-02-24:
From Netatmo response I have to concert myself all the values according to the user unit settings.

Will be available in the next app release.

Gulivertx, 2020-03-01:
Fixed with the last commit : https://github.com/Gulivertx/cbatmo/commit/890090b1064a4a87bf9bedf0dc63d54fae41f2bc

closed 2020-03-01 by Gulivertx