Gulivertx/cbatmo #10: Wind is showed in km/s even is changed to m/s

jukkaforss, 2020-02-24:
If I change unit from km/s to m/s, it changes km to ms, but numeric value is still km/s value.
Is this code bug, value not divided by 3.6?

jukkaforss, 2020-02-24:
[image]
[image]
Images of values

Gulivertx, 2020-02-24:
Hi same problem as https://github.com/Gulivertx/cbatmo/issues/9
I close this issue, please follow the issue 9 for the update.

closed 2020-02-24 by Gulivertx

Gulivertx, 2020-03-01:
Fixed in the last commit :  https://github.com/Gulivertx/cbatmo/commit/890090b1064a4a87bf9bedf0dc63d54fae41f2bc