Gulivertx/cbatmo #12: Dark Sky API Closed

Levanterman, 2020-04-11:
Hi 

I was going to make this awesome weather station you have created for my Netamo, i started and then immediately found that Dark Sky has been bought by Apple and the API is now closed to new subscribers. Existing subscribers are OK until 2021. More info here ->  ( [https://blog.darksky.net/](url) ) 

If I install cbatmo will i be able to edit the JS to parse and json weather report?

Thanks  - Nige.

Gulivertx, 2020-04-16:
Correct, Dark Sky is now a part of Apple and the JSON API is going to stop in the end of 2021. So for me and already registered user until the end of last year it's OK, but for you or any new user this is a problem to use Cbatmo...

So there is a need to move to another API, but currently I do not know which one is good, maybe Open Weather API is OK but I need to investigate. For now, I'm quite busy with another jobs then I will not investigate this problem right now but sure I will have a look asap.

For sure you can implement a new JSON API and if you can bind the data same as darksky you will not need to change anything to the frontend code. If you do that it will be great to create a pull request for your job :)

Levanterman, 2020-04-16:
Hi
That's great, I will attempt to make the conversion to the suggested API , and if I'm successful , for sure I will upload the finished code , I'll keep you informed , thanks. :-)

Gulivertx, 2020-04-16:
haha to migrate from darksky there is already a guide :)
https://openweathermap.org/darksky-openweather

Levanterman, 2020-04-16:
> haha to migrate from darksky there is already a guide :)
> https://openweathermap.org/darksky-openweather

Oh brilliant, that looks pretty straight forward :-) thank you

Gulivertx, 2020-04-26:
@Levanterman did you try something, should I help you and go further with this ?

Levanterman, 2020-04-26:
> @Levanterman did you try something, should I help you and go further with this ?

Hi :-) I created a 'OpenWeatherMap-Node' which retrieves the JSON weather, I am a .NET programmer but my Node js knowledge isn't great so i have been trying to find out how it works, with yarn ( which i can see hosts the dev and it works) . 
I edited the .env and i can see my netatmo modules, that all works great.

I have never used Express JS, but it looks like inside the Server.js , the APP communicates with the Darksky to get the json updates?

I added this to Server.js
```
const OpenWeatherMapHelper = require("openweathermap-node");
const helper = new OpenWeatherMapHelper(
	{
		APPID: 'xxxxxxxxxxx',
		units: "imperial"
	}
);
```

I think I need to replace this section with the OpenWeatherMap/node 

```
app.get('/darksky/:latitude/:longitude/:lang/:units', (req, res, next) => {
    if (!req.params.latitude ||
        !req.params.longitude ||
        !req.params.lang ||
        !req.params.units) return res.status(400).json({status: 'error', msg: 'Bad request'});

    DarkSkyApi.units = req.params.units; // default 'us'
    DarkSkyApi.language = req.params.lang; // default 'en'

    const position = {
        latitude: req.params.latitude,
        longitude: req.params.longitude
    };

    DarkSkyApi.loadItAll('hourly,flags', position)
        .then(result => {
            res.json(result)
            })  });
```

And then edit the  change the Dark Sky Parameters to match Openweather map parameters.
But I cant work out how to call the OpenWeatherMapNode from Server.js yet, assuming this is the correct approach. Any help would be great please, but I am slowly making progress.

:-) 

Thanks Merci

Gulivertx, 2020-04-27:
Hi, from my point of view I think we do not need any lib to query openweater api, I would prefer to do  a direct query from express route.

Correct about express, the server is created with server.js file.

So one proposition is to add a new route in server.js to handle an openweather query, and serve to JSON result to the React client.

Create a new route in express after the one for darksky for instance :
```javascript
....
app.get('/openweather/:latitude/:longitude/:lang/:units', (req, res, next) => {
    // We want to verify that each needed parameters are set in the request
    if (!req.params.latitude ||
        !req.params.longitude ||
        !req.params.lang ||
        !req.params.units) return res.status(400).json({status: 'error', msg: 'Bad request'});

    request.get(`https://api.openweathermap.org/data/2.5/onecall?lat=${latitude}&lon=${longitude}&units=${units}&lang=${lang}&appid=${process.env.OPENWEATHER_API_KEY}`, (error, result) => {
        // TODO in case of error throw an exception
        res.json(result)
    })
});
...
```
I didn't test it just do a rough example, to work in the .env file you have to add a new variable "OPENWEATHER_API_KEY", then we have also to verify if the parameters lang and unit works like this...

Gulivertx, 2020-04-27:
I did an implementation of the API in server.js. Please look the new new branch [openweather](https://github.com/Gulivertx/cbatmo/tree/openweather).
You can check the [last commit](https://github.com/Gulivertx/cbatmo/commit/f95963b4e675bc51fe4285a347727433d4ec4ae2) to see what I do.

The next step is to create for React/Redux a new model and new store for OpenWeather.

Gulivertx, 2020-04-27:
Still going ahead and finally do the binding of API in Redux store with a new model.
If you have redux dev tools installed you will see a new store call openweather with data coming from API.
[See last commit for code change](https://github.com/Gulivertx/cbatmo/commit/14ebc70ef7a9d63694655b6b951544c44f08f949)

Levanterman, 2020-04-28:
> Still going ahead and finally do the binding of API in Redux store with a new model.
> If you have redux dev tools installed you will see a new store call openweather with data coming from API.
> [See [...]

Thank you, i will continue to play and see if i can make progress :-)

Gulivertx, 2020-04-30:
UI now adapted to use OpenWeather. The fix is merged to the master branch ;)
https://github.com/Gulivertx/cbatmo/pull/15

closed 2020-04-30 by Gulivertx