Gulivertx/cbatmo #13: ADD MODULES

Snuoy, 2020-04-15:
hi everyone,

I'm just starting out, and I would just like to add 2 interior modules.
I can't find "Netatmo.js" in "client / components"

The photo shows what I want to do

Can you help me please?

The photo shows what I want to do
[image]

Gulivertx, 2020-04-16:
Currently CBatmo do not support more than one internal additional module.
If you want to add yours, their is a need to register them in the redux store.
This job must be done in the NetatmoNaMain model : https://github.com/Gulivertx/cbatmo/blob/master/client/models/NetatmoNAMain.ts#L6-L12
https://github.com/Gulivertx/cbatmo/blob/master/client/models/NetatmoNAMain.ts#L59-L64
https://github.com/Gulivertx/cbatmo/blob/master/client/models/NetatmoNAMain.ts#L136-L169

And then creating new components / containers and add them here
https://github.com/Gulivertx/cbatmo/blob/master/client/components/App.tsx#L38-L95

Snuoy, 2020-04-16:
thank you very much for the answer.
it's hot for me especially with the redux store, so I give up.

thanks again

Gulivertx, 2020-04-16:
if you want I can do the part for the redux store (I need your help to know what Netatmo give for modules name) and then next you will have to play with UI to show your 3 indoor module.
Sounds OK for you ?

Snuoy, 2020-04-17:
ohhh yessss !!!
Thank you !!!
My 3 modules are:
- Ines Room
- Parents
- Office

That's right ?

Gulivertx, 2020-04-25:
No that :), this is the name you give to your modules, what I need is the module type given by Netatmo which is unique for each module. Hi guess it should be :

- NAModule4 = First additional indoor module
- NAModule5 = Second additional indoor module
- NAModule6 = Third additional indoor module

I just need a confirmation for the second and third to be sure before to add them.

Could you please verify because I don't have this devices? It is quite easy if you have a CBatmo running installation.
Open your browser (I will explain for Google Chrome) and reach your Raspberry Pi where CBatmo is running, i.e for mine : http://10.0.0.143:3000/
Then open the dev tool (it is call "Inspect" if you right click some where), then in the DEV tool go to **Network** tab and select the query to get station data. If you do not have any request just refresh the page.
Finally to found the type follow the instruction on my screenshot ;)

[image]

Snuoy, 2020-04-25:
oops désolé ;-) 

voici ce que tu veux :
[image]

ils ont tous les trois le même nom , :-(

thanks again

Gulivertx, 2020-04-26:
Parfait merci.
I'm gonna implement them !

Gulivertx, 2020-04-26:
Ok I did an update where you should have available in the Redux store all indoor modules.
Please check the [last commit](https://github.com/Gulivertx/cbatmo/commit/53e46417db1bbaebbb831ecf1decc8ba7ae262b0) for more information about the change.

Then to check if every thing is correct, update your local repo with a `git pull` then please install the last npm packages because I also did an update, just run `yarn install`.
Finally build and go in your browser to check if your received correctly the data (I did it blindly). You have to open the Google Chrome dev tools and display the debug log message to see an entry with a name as "NetatmoNAMain", open the modules section and you should have modules with name :

- indoor
- indoor_second
- indoor_third

Please could you confirm that for each of these modules you have all data available.

Snuoy, 2020-04-28:
Hi,

Thanks again for everything you do,

I don't know if the image is what you want?

[image]

Snuoy, 2020-04-28:
How are the three modules integrated on the page?

Gulivertx, 2020-04-29:
Yep correct ! Thx. So there is a problem with the INDOOR_THIRD which not get any data, I will check it.

To integrate them you have to create two new containers by copy/past this file : https://github.com/Gulivertx/cbatmo/blob/master/client/containers/ModuleNetatmoIndoorContainer.ts

Create a new file call ModuleNetatmoIndoorSecondContainer.ts and one other call ModuleNetatmoIndoorThirdContainer.ts

Then edit them and change this : https://github.com/Gulivertx/cbatmo/blob/master/client/containers/ModuleNetatmoIndoorContainer.ts#L8 like :
```javascript
module_data: netatmo.station_data?.modules.INDOOR_SECOND
and
module_data: netatmo.station_data?.modules.INDOOR_THIRD
```

Then add the new container to App.tsx (this is the main view) file : https://github.com/Gulivertx/cbatmo/blob/master/client/components/App.tsx

You have to play in this area to add them :
https://github.com/Gulivertx/cbatmo/blob/master/client/components/App.tsx#L38-L95

I'm thinking to create a new behavior in the app, like when the app start and detect available modules, add a new choice for the user to choose a specific layout according to its available modules, but not yet no time to do it and it's quiet a big job...

Gulivertx, 2020-04-29:
Ok found and make the correction for the INDOOR_THIRD module not getting any data, just do a `git pull` in the root project folder to get the modification

Snuoy, 2020-04-29:
Hi chef,

THANK YOU,

It's perfect,

[image]

Gulivertx, 2020-04-30:
Nice job!
Luneville, c’est ou ça, ça existe vraiment? 🤔

Snuoy, 2020-04-30:
Si Si ça existe vraiment ;-) , à 20 km de Nancy, et toi tu es de Charleville-Mézières  ?

Good Idea for the issue open ;-)

I have a little question, what is yarn version on rasbian should be used ?

Gulivertx, 2020-05-01:
Non, je suis en Suisse ;)
It doesn't matter about the yarn version, why ?

Snuoy, 2020-05-01:
It's good for yarn, i had problem with nodejs update.

Currently, i have a problem with the execution of the script

"code = exited, status = 1 / FAILURE"

I look for ...

[image]

Snuoy, 2020-05-01:
ahh its working, I don't understand, I didn't do anything

[image]

Snuoy, 2020-05-01:
I don't understand, it alternates
[image]

Snuoy, 2020-05-01:
[image]

I think it comes from me,

Gulivertx, 2020-05-02:
ok, yarn is installed ?
Could you locate yarn ? `whereis yarn`

Did you also edited the service script before to deploy ?
```bash
# SystemD example script
# This script is used on my RPI installed with ArchLinux
# Modify the path of cbatmo folder in WorkingDirectory option
# Modify the User and group with your username and group
# Don't forget to install yarn on your RPI

[Unit]
Description=CBatmo WEB server
# Set dependencies to other services (optional)
#After=mongodb.service

[Service]
# Start the js-file starting the express server
ExecStart=/usr/bin/yarn start
WorkingDirectory=/home/alarm/cbatmo
Restart=always
RestartSec=10
StandardOutput=syslog
StandardError=syslog
SyslogIdentifier=CBatmo
# Change to a non-root user (optional, but recommended)
User=alarm
Group=alarm
# Set environment options
Environment=NODE_ENV=production PORT=3000

[Install]
WantedBy=multi-user.target
```

you have to change  yarn path and cbatmo path,these lines
ExecStart=/usr/bin/yarn start
WorkingDirectory=/home/alarm/cbatmo

And also change user and group, for raspbian you should have **pi** for both.

Snuoy, 2020-05-02:
yes yarn is installed,  its located in "**usr/bin/yarn /usr/share/yarn**"

yes, i have edited the service script before deploy, in this one i put : 
- ExecStart=**/usr/bin/yarn start**
- WorkingDirectory=**/home/pi/cbatmo**
-user and group : **pi**

Snuoy, 2020-05-02:
I just found, i deleded start in path "ExecStart=/usr/bin/yarn **start**"

[image]

Snuoy, 2020-05-02:
now i have an other problem :-(
the service  constantly restarts

[image]

Snuoy, 2020-05-02:
it turns lol,

I just write "start" in the path "ExecStart=/usr/bin/yarn **start**"

now I will play to change the resolution because 10 "screen

ps: from my computer i hav not access to 10.0.0.10:300 

ENCORE MERCI CEDRIC, tu devrais postuler chez google, pour améliorer leurs systèmes , ;-) 
You are the BEST

Gulivertx, 2020-05-02:
Nice! 
http://10.0.0.10:3000 and not http://10.0.010:300

Haha! Thanks 😊 

For your screen resolution problem I have the same. I would like to use the app on my iPad and iPhone with public access from outside my home (i’m gonna change the behavior of the loading app to ask netatmo login and password when the current device is not already registered). The thing is that I did a fix width and height and not responsive design, which is bad...

Then here is what I’m going to modify:
- Netatmo login if device is new
- Responsive design to adapt UI according to the screen
- Add service worker to use the app as a PWA

Snuoy, 2020-05-02:
You also have the second column which is belowthe first column ? 

[image]

Gulivertx, 2020-05-02:
should not be below, this what I have.
To continue your dev my advise is to switch to the new branch [2.3.x](https://github.com/Gulivertx/cbatmo/tree/2.3.x).
I started integration for responsive layout, I first added a login step if no Netatmo refresh_token already save in localstorage.
I also started to use [reflexbox](https://rebassjs.org/reflexbox/) for the responsive design which will help a lot to get something correct. I think, if you want to play with the design, this branch  will be better because it will will be in the continuity of my current work. Good luck !

Snuoy, 2020-05-03:
OK THANK, i trying reflexbox.

good luck

Gulivertx, 2020-05-07:
Hi @Snuoy , I'm currently developing a version 2.3.0 of CBatmo with the support of 3 inside modules (I bought them), if you follow the branch 2.3.x and read the readme you will have an overview of the next step. CYA!

Snuoy, 2020-05-09:
hi,

 Cool for your 3 modules and good luck with the dev

Thanks

exoticatom, 2020-06-04:
hello,
once again thanks for great work.
I have main module, 2 in rooms and one outdoor.
Would this cover Main Station + 3 inside room modules? 
I am missing one.

[image]

exoticatom, 2020-06-04:
hello,
once again thanks for the great work.
I have installed, 2.3.1 version.

I saw that you mentioned that "Currently CBatmo do not support more than one internal additional module." Is this still the case?

As a response I've got:

"type":"NAModule1",
                  "module_name":"Outdoor",

 "type":"NAModule4",
                  "module_name":"Sleeping",

  "type":"NAModule4",
                  "module_name":"Working",
Finally, I am not seeing internal module NAModule4 "working" on the graphs as you confirmed.

Would it be possible to add that module too ? I am going trough your post,but I am unsure if this solution is applicable for me too ?

exoticatom, 2020-06-04:
I am so far that just need to edit App.tsx,  but I have no clue how to :

```
You have to play in this area to add them :
https://github.com/Gulivertx/cbatmo/blob/master/client/components/App.tsx#L38-L95

```

exoticatom, 2020-06-04:
what I found out that it works out of the box for mobile, just not on chromium-browser on touch screen connected to Raspi.

exoticatom, 2020-06-04:
last question :-) if I change App.tsx, do I need to make again redeployment to raspi?

exoticatom, 2020-06-04:
never mind I managed to change App build it again and get my last module :-). Cheers

exoticatom, 2020-06-04:
[image]

exoticatom, 2020-06-04:
[image]

tdostler, 2020-06-04:
> never mind I managed to change App build it again and get my last module :-). Cheers

Can you explain in some steps what you did?

exoticatom, 2020-06-04:
Hi,
First of all I followed the posts here. However Cedric already included change with version 2.3.1 and he mentioned he  is working on it. This what I did is only temporary until there is a way to add all your modules. And solution below is inflexible. You will have to experiment for your setup. 

 First of all posts before are misleading because they are related to old version of the app and file changed  https://github.com/Gulivertx/cbatmo/issues/13#issuecomment-621018081
Lines form this message are not anymore correct. This took me long time to figure out :-).

Anyways. You need to change App.tsx.
So what is missing is _only_ visual setup which is defined in this file `cbatmo/client/components/App.tsx`

Steps:
1. All your changes you have to do where you have cloned git. then you have to build another deployment. 
2. Since I dont have weather and wind module, I decided to change in App.tsx following lines:
https://github.com/Gulivertx/cbatmo/blob/master/client/components/App.tsx#L124-136

from:

```
                           {
                                available_modules?.RAIN ? (
                                    <Box width={[ '100%', '50%' ]}>
                                        <ModuleNetatmoRainContainer />
                                    </Box>
                                ) : null
                            }
                            {
                                available_modules?.WIND ? (
                                    <Box width={[ '100%', '50%' ]}>
                                        <ModuleNetatmoWindContainer />
                                    </Box>
                                ) : null
                            }
                            <Box width={[ '100%', '50%' ]}>
                                <ModuleNetatmoGraphContainer />
                            </Box>

```
to:

Follow the logic, remove wind and rain, and change to 100% graph
```

                           {
                                 available_modules?.INDOOR_SECOND ? (
                                    <Box width={[ '100%', '50%' ]}>
                                        <ModuleNetatmoIndoorSecondContainer module_name='indoor_second' />
                                    </Box>
                                ) : null
                            }
                            {
                                available_modules?.INDOOR_THIRD ? (
                                    <Box width={[ '100%', '50%' ]}>
										<ModuleNetatmoIndoorThirdContainer module_name='indoor_third' />
                                    </Box>
                                ) : null
                            }
                            <Box width={[ '100%', '100%' ]}>
                                <ModuleNetatmoGraphContainer />
                            </Box>

```

3. rebuild `yarn run build`
4. stop the service
5. deploy the package again. (I use manual zip transfer to PI) Explained : https://github.com/Gulivertx/cbatmo#create-an-archive
5. restart the service.

Snuoy, 2020-06-04:
hi,

Indeed, I also found that version 2.3.1 does not add the additional modules ...

Gulivertx, 2020-06-05:
@exoticatom thanks to share your experience.
Currently the indoor modules second and third are not implemented, **but only in the view**, I already prepare everything but did not find the time to code the implementation in the view.
The thing is it's difficult to have view which will work for every combinaison of possible available modules. The second thing is that you have to think about that I do this project during my free time, and I have a lot of other thing to maintain in this project, then please be patient, it will come.

The good thing is that I now have a good state to with the last two releases to implement this dynamic view according to the users available modules. And I finally found how I want to code this!

@exoticatom to answer you, yes if you change the code you have to rebuild the project and redeploy it.
And it is also correct for mobile view, the portrait mobile view already include everything, only the portrait view is still not implement this.

Levanterman, 2020-06-07:
Hi, this thread has been interesting to read. After the weather API change last month, CBATMO has been working perfectly thank you. But I have an out door module, rain module, and two indoor modules but no wind module, so i have one empty box in CBATMO. Is the above code changes enough for me to add the indoor module that is not showing up in CBATMO, which will fill the empty box? Thanks

exoticatom, 2020-06-07:
yes. that's exactly what's about. You can wait for official release or try some of the stuff explained in last posts of this thread.

Levanterman, 2020-06-07:
> yes. that's exactly what's about. You can wait for official release or try some of the stuff explained in last posts of this thread.

Thanks :-)

Gulivertx, 2020-07-01:
This issue is not yet cover by the the last 2.3.2 release. There is more modules layouts which will work out of the box but still not all modules possible mixing. Jeep calm everything will come in the next release.

Levanterman, 2020-11-28:
Hi

In v.2.3.3 does the above method still work to add more modules. I have 3 indoor modules and rain module  (no wind module)

Thanks

Nige.

Levanterman, 2020-12-14:
> Hi
> 
> In v.2.3.3 does the above method still work to add more modules. I have 3 indoor modules and rain module (no wind module)
> 
> Thanks
> 
> Nige.

The information above was great to be able to add another indoor module and re-arrange the GUI - thanks a lot. Let me know if you'd like me to upload the code back. 

[image]

closed 2022-02-17 by Gulivertx