Gulivertx/cbatmo #14: Is there a way I can translate it to English?

AndyMansson, 2020-04-29:
Hello Gulivertx, first I just wanted to say your project is the answer to my wishes.
Been looking for a way to display my Netatmo data in a nice way. And this is defiantly it.
Thank you.

I have had your previous version (2.0.0) for a while on my Mac, but will now go to Raspberry Pi and I saw that you have updated it. I would definitely like to use the nice new version.

As I read in your readme, you have plans to make it possible to have your application in English,
but it is not yet so. Is this something I could change myself? Or can I contribute somehow? (Not the best coder though...)

Thank you for a great application. And a great instruction for us new with Linux/Raspberry Pi.

Gulivertx, 2020-04-29:
Hi, yep it is still in the pipeline to have En/Fr support. This is not a huge job, so I can do it in few days.

AndyMansson, 2020-04-29:
Looking forward to that. Thank you. 🥇

Gulivertx, 2020-04-30:
I re-open this issue and will close it when the job will be done.

closed 2020-05-01 by Gulivertx