Gulivertx/cbatmo #25: APP_SECRET=

ocvelbye, 2020-05-10:
Where do I find this information?

Thanks.

Gulivertx, 2020-05-11:
This is like a second password that you choose. Can be what you want.

AndyMansson, 2020-05-18:
Hello Gulivertx, is there still a way to have automatic login as soon you access the webpage?
Ie. not use this new functionality.

As I also display my data embedded on my personal (but official) webpage.
And people without my login won't be able to see it.

Thanks again for a great application.

Gulivertx, 2020-05-18:
How do you embed this? With an iframe?
I will not put back auto-login, sorry, but I do not want anyone can use the app. Openweather api has a quota of maximum request and when you reach them you have to subscribe to a non free plan. 

If you want auto-login you will have to implement yourself. Of course you can still use the previous version Of cbatmo or implement the old code without login page in the current version.

AndyMansson, 2020-05-18:
Yes it is in a iframe :) Have a look if you want http://odahl.net/weather/
Well 60 calls / minute for the free plan seems a lot, at least for my small private site.

Then I will stick with the version 2.2.1 since I don't know how to change the code.
Thank you for the quick answer.

closed 2020-05-23 by Gulivertx