Gulivertx/cbatmo #26: Scrollbar I.

jukkaforss, 2020-05-17:


jukkaforss, 2020-05-17:
[image]

jukkaforss, 2020-05-17:
Seems that screen is bit bigger than 800x600 in 2.30 version.

Gulivertx, 2020-05-17:
Will be fixed soon in version 2.3.1.

karba71, 2020-06-02:
Hello,

since we have any rain today i have this fidget scrollbars on my screen.
I have V2.3.1 installed.

[20200602_194910_Trim.zip](https://github.com/Gulivertx/cbatmo/files/4719767/20200602_194910_Trim.zip)

Gulivertx, 2020-06-03:
Ho damn ! The scrollbars appears only when Rain graph as some data to render ? When there was no Rain there is no scrollbar ?
Also, your app is really shaking or is it coming from the video ?

jukkaforss, 2020-06-03:
Mine is acting same, really shaking with scrollbars. 

-Jukka-

Send from iPhone

karba71, 2020-06-03:
Hello,

its really shaking.
And i think it comes from the rain.
If we have no rain then the data in 0 - 1 digit.
If we have rain or no data (NAN) then we have 3 digit (0,6 or NAN).
And then we have this scrollbars and shaking.

I use Chromium V 78.0.3904.108

[image]

jukkaforss, 2020-06-03:
I also use Chromium V 78.0.3904.108.
When this happening, if I killall chromium and restart it, it will be fine for awhile.
Here is video.
[IMG_7375.zip](https://github.com/Gulivertx/cbatmo/files/4721957/IMG_7375.zip)

Gulivertx, 2020-06-03:
Ok thanks for your reports, I don’t have this trouble but I use webkit-gtk. 
I also see on your screenshots that the padding for lot of elements are not correct with chromium. I’m gonna investigate this...

Gulivertx, 2020-06-05:
OK found the problem about the rain graph, the responsive chart is not re-render if the right values change. I will implement a render refresh of the graph each time the rain values change to force the graph re-render with the available width which should fix this shaking issue.

tdostler, 2020-06-28:
> OK found the problem about the rain graph, the responsive chart is not re-render if the right values change. I will implement a render refresh of the graph each time the rain values change to force the graph re-render with the available width which should fix this shaking issue.

Can you explain, and give some hints, how to re-render the rain-graphic?

Gulivertx, 2020-06-30:
@tdostler to force React to re-render any components there are different ways, the most common is to change a value in the component state or props. But there is also the ability to add a key to a component and by updating the key value React will re-render the component (not re-render for this case but unmount  and mount again the component), the last method is to use forceUpdate method (https://reactjs.org/docs/react-component.html#forceupdate).

I already tried to do it in the branch deploy_script, by adding a key on the graph component where the key has value of the rain module. So when the rain values change, the key of the graph get this new values, then the graph is unmount and mount again, but unfortunately it is not working, here is the change of my code : https://github.com/Gulivertx/cbatmo/commit/fc7beb8544d09d86e9f5a1833adb9783665c387c#diff-57f2be8e0124c6dfd7519e16ebadfed1R22-R28
I think I add the key id bad component because it's on the chart itself, which is re-render correctly but do not know his available width, I will try to re-render the parent component of the chart to see if it make any improvement...

Gulivertx, 2020-07-03:
Damn it!!!!
Still shaking. I’m gonna make an update this weekend for I hope definitively suppress this issue. 
Related issue : https://github.com/Gulivertx/cbatmo/issues/41
Thank’s @karba71 for the report

PR #42 (2020-07-03): Fix issues 26 and 41
fix #26 
fix #41
closed 2020-07-03 by Gulivertx