Gulivertx/cbatmo #28: problem with yarn install --production

karba71, 2020-05-25:
Hello,

thanks for update this project!!!

I create the archive cbatmo to my raspi and unzip.
Then i get this:

/home/pi/cbatmo$ yarn install --production
-bash: Zeile 10: yarn: Kommando nicht gefunden.

What is the Problem?

Gulivertx, 2020-05-25:
Is yarn installed in your system?

karba71, 2020-05-25:
Hello Cedric,I have yarn installed on my PC.How can i install yarn on my Raspi?KarbaVon meinem Samsung Galaxy Smartphone gesendet.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/Gulivertx/cbatmo/issues/28#issuecomment-633617536",
"url": "https://github.com/Gulivertx/cbatmo/issues/28#issuecomment-633617536",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

karba71, 2020-05-25:
Hello Cedric,

i think i have install yarn on my raspi.
I do this directly on the raspi, not over ssh.

And now i have this result:

/home/pi/cbatmo$ systemctl status cbatmo.service
● cbatmo.service - CBatmo WEB server
   Loaded: loaded (/etc/systemd/system/cbatmo.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Mon 2020-05-25 23:11:47 CEST; 4s ago
  Process: 897 ExecStart=/usr/bin/yarn start (code=exited, status=217/USER)
 Main PID: 897 (code=exited, status=217/USER)

karba71, 2020-05-26:
Hello,

I have fixed my mistake within rpi/cbatmo.service.
User and group are pi now.

[Service]
# Start the js-file starting the express server
ExecStart=/usr/bin/yarn start
WorkingDirectory=/home/pi/cbatmo
Restart=always
RestartSec=10
StandardOutput=syslog
StandardError=syslog
SyslogIdentifier=CBatmo
# Change to a non-root user (optional, but recommended)
User=pi
Group=pi
# Set environment options
Environment=NODE_ENV=production PORT=3000

And the result:
/home/pi/cbatmo$ systemctl status cbatmo.service
● cbatmo.service - CBatmo WEB server
   Loaded: loaded (/etc/systemd/system/cbatmo.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Tue 2020-05-26 19:59:27 CEST; 6s ago
  Process: 1876 ExecStart=/usr/bin/yarn start (code=exited, status=203/EXEC)
 Main PID: 1876 (code=exited, status=203/EXEC)

What is the Problem?

THX karba

Gulivertx, 2020-05-26:
ok, could be a lot of different things...

Please try to run the server from command line without using systemd now.
Juste go to your cbatmo folder then `yarn run start`. 
What is the result of this command ?

karba71, 2020-05-26:
Hello Cedric,

 

i have found another mistake.

Change the first variable APP_ENV=dev to APP_ENV=prod.

 

yarn run start

yarn run v1.22.4

$ node server.js

Server running on http://localhost:3000 as production

 

karba

 

Just go to your cbatmo folder then yarn run start.

karba71, 2020-05-26:
When i check http://localhost:3000 on my raspi:

Failed to lookup view "index" in views directory "home/pi/cbatmo/views"

This Directory only contains a file error.ejs

Gulivertx, 2020-05-27:
This is because there is no build, unfortunately you can not build the app from a raspberry, node-sass is not compatible with arm processor (I did not check if it is still the case but I assume that it is). 
You must build the app from a x86 / x64 processor has intel or amd. 

Please follow the guide in the Readme to build your app and at the end copy the build result, everything should be well explain in the readme. 

The best way if you are not confortable to build and copy from another computer is to use a MacOS or Linux system and the deploy script in the rpi folder for an auto deployment. Good luck.

karba71, 2020-05-28:
Hello,

i've got it:

/home/pi/cbatmo$ sudo cp rpi/cbatmo.service /etc/systemd/system/
/home/pi/cbatmo$ sudo systemctl enable cbatmo.service
/home/pi/cbatmo$ sudo systemctl restart cbatmo.service
/home/pi/cbatmo$ systemctl status cbatmo.service
● cbatmo.service - CBatmo WEB server
   Loaded: loaded (/etc/systemd/system/cbatmo.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2020-05-28 02:09:36 CEST; 16s ago
 Main PID: 4755 (node)
    Tasks: 23 (limit: 2200)
   Memory: 61.6M
   CGroup: /system.slice/cbatmo.service
           ├─4755 node /usr/share/yarn/bin/yarn.js start
           ├─4779 /bin/sh -c node server.js
           └─4780 /usr/bin/node server.js

Mai 28 02:09:36 raspberrypi systemd[1]: Started CBatmo WEB server.
Mai 28 02:09:40 raspberrypi CBatmo[4755]: yarn run v1.22.4
Mai 28 02:09:40 raspberrypi CBatmo[4755]: $ node server.js
Mai 28 02:09:45 raspberrypi CBatmo[4755]: Server running on http://localhost:3000 as production

cbatmo is running, but how can i login?

???? my mail-adress
???? my passwort
???? ????

Gulivertx, 2020-05-28:
Nice! The last field is your app secret that you specified in your .env

karba71, 2020-05-28:
Hello,

 

I had sucess with a VW on my PC last night!!!

 

Bud I can’t login!

 

1: my Mail-Adress or Client-ID for Netatmo?

 

2: my Password for Netatmo or Client secret?

 

3: App secret (i leave it empty in .env?)

 

On my Raspi i can see the login-Window.

But i can’t see this window on my pc.

But i can see the cbatmo-background.

 

karba

Gulivertx, 2020-05-28:
First: app_secret in .env cannot by empty, I do not allow null value, sorry, so on your Raspi specify a secret key, what you want, then restart the server from systemctl. 
After that it should works. 

Second: For your windows problem, could you please tell me what is your web browser and the version.

karba71, 2020-05-29:
It‘s running!!!

 

Next step is Kiosk-Mode – i hope without screensaver and any notification and with auto-login.

 

I have test with IE11.

I can see only the background from cbatmo.

 

Then i testet with Edge – it’s running.

Gulivertx, 2020-05-29:
Ha yes I do not support this IE11 dinosaur 🙂 which should not by used anymore!

For kiosk mode in the rpi folder there is script for what you want running on Weston / Wayland.

karba71, 2020-05-29:
Hello Cedric,

 

I will check Weston / wayland with your script.

 

1 question:

 

How can i change the data at the graph at the right-side?

The content is changing and i dont know why.

 

1 whish:

 

Can you integrate this status-informations?

 

 

Or in this way:

 

 

Thanks and good night

 

karba

exoticatom, 2020-06-02:
Hi, i tried installing from windows to PI (2020-05-27-raspios-buster-lite-armhf)  but deploy.sh fails with error:

file is downloaded **cbatmo-2.3.1.tar.gz**  but
1) after download script tar fails with an error:
```
gzip: stdin: unexpected end of file
tar: Child returned status 1
tar: Error is not recoverable: exiting now
-bash: line 3: cd: cbatmo: No such file or directory
```

2) if I clone package directly to PI, try to follow deploy script by executing:
```
yarn install --production
Usage: yarn [options]

yarn: error: no such option: --production

Could you please help, I have no clue what I am doing wrong.

```
thank you

Gulivertx, 2020-06-02:
changing the data in graph is just by clicking any values.

you said : Can you integrate this status-informations?
What are you talking about ? There is no pictures in your message.

closed 2020-06-02 by Gulivertx

karba71, 2020-06-02:
Hello Cedric,

changing the data in graph:
I can take data from the 2. indoor-module (temp, CO2, acoustic), the the outdoor-module (temp, humidity, pressure), the rain-sensor and the wind-sensor.

But, if i take any data from the main-module cbatmo breakes.

The graph will store which data i select, but not the time (12 hour, 1 day or 1 month).
It starts every time with 12 hour.

I think at the rain-gauge is a unit slips:

[image]

karba

Gulivertx, 2020-06-03:
Could you open a new issue for the rain unit?