Gulivertx/cbatmo #29: Help needed for instllation

exoticatom, 2020-06-02:
Hi there,
I tried everything to install, but always getting an error immediately  after deploy has been exectued:
```
mihajloz@DESKTOP-9R6JH8T:~/cbatmo/rpi$ ./deploy.sh
CBATMO DEPLOY SCRIPT
CBATMO VERSION 2.3.1
______________________________________________________

Build cbatmo for production
**00h00m00s 0/0: : ERROR: [Errno 2] No such file or directory: 'run'**

Start packaging version: 2.3.1
Generated archive with name cbatmo-2.3.1.tar.gz

Do you want to upload this archive into your Raspberry Pi? yes
Please specify your Raspberry Pi IP address: 192.168.168.165
Please specify your Raspberry Pi username: pi
Upload cbatmo-2.3.1.tar.gz on your Raspberry Pi 192.168.168.165
pi@192.168.168.165's password:
Connected to pi@192.168.168.165.
sftp>   put cbatmo-2.3.1.tar.gz
Uploading cbatmo-2.3.1.tar.gz to /home/pi/cbatmo-2.3.1.tar.gz
cbatmo-2.3.1.tar.gz                                                          100%  594KB   2.0MB/s   00:00
sftp>   quit

Do you want to install cbatmo into your Raspberry Pi? yes
Install CBatmo on your Raspberry Pi 192.168.168.165
Pseudo-terminal will not be allocated because stdin is not a terminal.
pi@192.168.168.165's password:
Linux raspberrypi 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
**Usage: yarn [options]

yarn: error: no such option: --production**
Created symlink /etc/systemd/system/multi-user.target.wants/cbatmo.service → /etc/systemd/system/cbatmo.service.
```

What do you think where could be an error ?
What am I missing?

exoticatom, 2020-06-02:
if I execute form cbatmo dir:
```
pi@raspberrypi:~/cbatmo/cbatmo $ yarn run start
00h00m00s 0/0: : ERROR: [Errno 2] No such file or directory: 'run'
pi@raspberrypi:~/cbatmo/cbatmo $ yarn --version
0.32+git
```

exoticatom, 2020-06-02:
if I execute form cbatmo dir:

```
pi@raspberrypi:~/cbatmo/cbatmo $ yarn run start
00h00m00s 0/0: : ERROR: [Errno 2] No such file or directory: 'run'
pi@raspberrypi:~/cbatmo/cbatmo $ yarn --version
0.32+git

pi@raspberrypi:~/cbatmo/cbatmo $ ls -al
total 396
drwxr-xr-x 5 pi pi   4096 Jun  2 18:10 .
drwxr-xr-x 6 pi pi   4096 Jun  2 18:12 ..
-rw-r--r-- 1 pi pi    453 Jun  2 18:10 .env.dist
-rw-r--r-- 1 pi pi    361 Jun  2 18:10 .gitignore
-rw-r--r-- 1 pi pi   1069 Jun  2 18:10 LICENSE
-rw-r--r-- 1 pi pi   2631 Jun  2 18:10 package.json
drwxr-xr-x 2 pi pi   4096 Jun  2 18:10 public
-rw-r--r-- 1 pi pi  10198 Jun  2 18:10 README.md
drwxr-xr-x 2 pi pi   4096 Jun  2 18:10 rpi
-rw-r--r-- 1 pi pi   7386 Jun  2 18:10 server.js
-rw-r--r-- 1 pi pi    663 Jun  2 18:10 tsconfig.json
drwxr-xr-x 2 pi pi   4096 Jun  2 18:10 views
-rw-r--r-- 1 pi pi   4796 Jun  2 18:10 webpack.config.js
-rw-r--r-- 1 pi pi 333944 Jun  2 18:10 yarn.lock

```

exoticatom, 2020-06-02:
if I execute form cbatmo dir:

pi@raspberrypi:~/cbatmo/cbatmo $ yarn run start
00h00m00s 0/0: : ERROR: [Errno 2] No such file or directory: 'run'
pi@raspberrypi:~/cbatmo/cbatmo $ yarn --version
0.32+git
pi@raspberrypi:~/cbatmo/cbatmo $ ls -al
total 400
drwxr-xr-x 5 pi pi   4096 Jun  2 18:47 .
drwxr-xr-x 6 pi pi   4096 Jun  2 18:12 ..
-rw-r--r-- 1 pi pi    481 Jun  2 18:47 .env
-rw-r--r-- 1 pi pi    453 Jun  2 18:10 .env.dist
-rw-r--r-- 1 pi pi    361 Jun  2 18:10 .gitignore
-rw-r--r-- 1 pi pi   1069 Jun  2 18:10 LICENSE
-rw-r--r-- 1 pi pi   2631 Jun  2 18:10 package.json
drwxr-xr-x 2 pi pi   4096 Jun  2 18:10 public
-rw-r--r-- 1 pi pi  10198 Jun  2 18:10 README.md
drwxr-xr-x 2 pi pi   4096 Jun  2 18:48 rpi
-rw-r--r-- 1 pi pi   7386 Jun  2 18:10 server.js
-rw-r--r-- 1 pi pi    663 Jun  2 18:10 tsconfig.json
drwxr-xr-x 2 pi pi   4096 Jun  2 18:10 views
-rw-r--r-- 1 pi pi   4796 Jun  2 18:10 webpack.config.js
-rw-r--r-- 1 pi pi 333944 Jun  2 18:10 yarn.lock

Gulivertx, 2020-06-02:
please update your yarn, this version is too OLD

```bash
[~] yarn -v
1.22.4
```

exoticatom, 2020-06-02:
Hi, 

Thank you for quick answer. 

Its absurd. I have installed latest raspi os 2020-05-27-raspios-buster-lite-armhf
I did `apt-get update / upgrade `and that was the version I've got - old version. 

I needed to use : 

`"curl --compressed -o- -L https://yarnpkg.com/install.sh | bash"`

```
pi@raspberrypi:~ $ yarn --version
1.22.4

```

I am going to next step now :-) . Fighting with requirement like webpack. 

Now missing:
```
./deploy.sh
CBATMO DEPLOY SCRIPT
CBATMO VERSION 2.3.1
______________________________________________________

Build cbatmo for production
yarn run v1.22.4
$ webpack
internal/modules/cjs/loader.js:638
    throw err;
    ^

Error: Cannot find module 'webpack-shell-plugin'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:636:15)
    at Function.Module._load (internal/modules/cjs/loader.js:562:25)
    at Module.require (internal/modules/cjs/loader.js:692:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at Object.<anonymous> (/home/pi/cbatmo_inst/webpack.config.js:3:28)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.

```

Is this all because I used light version? Darn...

I will install full version of PI OS tomorrow and try again, it's getting late here...cant think anymore.

Gulivertx, 2020-06-03:
No this is not about your os light version. First I think you did not install dependencies with the command yarn install. 
Second the build process not work on rpi and must be done in a x86 our x64 computer. The raspberry has a arm processor which is not compatible with some dependencies used during the build process. 

If you wait the next release I will push a new easier install process directly from rpi.

exoticatom, 2020-06-03:
ok thank you. This wasnt clear for me, if I can do it directly form PI.
I will try that next.

exoticatom, 2020-06-03:
May I ask if dependencies are needed on both computers? On pi and on x86?

exoticatom, 2020-06-03:
Ok thank you very much.  I will try that next. 

May I ask where are dependencies needed, on both pi and x86?

exoticatom, 2020-06-03:
sorry, I am stupid :-) reading instructions again.

Gulivertx, 2020-06-03:
There is dev dependencies which are installed with the command yarn or yarn install. 

There is production dependencies which are necessary to run the web server and only needed in rpi, installed with yarn install —production

exoticatom, 2020-06-03:
Finally.

```
info fsevents@2.1.3: The platform "linux" is incompatible with this module.
info "fsevents@2.1.3" is an optional dependency and failed compatibility check. Excluding it from installation.
info fsevents@1.2.13: The platform "linux" is incompatible with this module.
info "fsevents@1.2.13" is an optional dependency and failed compatibility check. Excluding it from installation.
[3/4] Linking dependencies...
[4/4] Building fresh packages...
Done in 226.46s.
● cbatmo.service - CBatmo WEB server
   Loaded: loaded (/etc/systemd/system/cbatmo.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2020-06-03 10:13:23 BST; 2s ago
 Main PID: 5451 (node)
    Tasks: 6 (limit: 2200)
   Memory: 3.8M
   CGroup: /system.slice/cbatmo.service
           └─5451 node /usr/share/yarn/bin/yarn.js start

```

now as advice if you can be more precise saying that yarn is needed on both side.
That it's a  must to execute deployment remotely.
TO get right version of yarn:

Installing Yarn on Debian

```
curl -sS https://dl.yarnpkg.com/debian/pubkey.gpg | sudo apt-key add -
echo "deb https://dl.yarnpkg.com/debian/ stable main" | sudo tee /etc/apt/sources.list.d/yarn.list
sudo apt update
sudo apt install yarn
```
If you already don’t have Node.js installed on your system, the command above will install it. Those who are using nvm can skip the Node.js installation with:

Verify the installation by printing the Yarn version number:

`yarn --version`

and the follow the steps as you described in readme. 

Important is "apt-get install yarn" -> will get you old version even if you did apt-get update/upgrade.

exoticatom, 2020-06-03:
:-) now if I may ask which credentials  ( username and password) shall I use to login?
APP_SECRET is clear, I have defined that

exoticatom, 2020-06-03:
Finally everything worked. But I can't login using my secret key and netatmo credentials. 

Could you please help ? what are credentials to be used?

exoticatom, 2020-06-03:
this is what I see in syslog - not sure if they are related:

Jun  3 12:59:58 raspberrypi CBatmo[4223]: { StatusCodeError: 400 - {"error":"invalid_client"}
Jun  3 12:59:58 raspberrypi CBatmo[4223]:     at new StatusCodeError (/home/pi/cbatmo/node_modules/request-promise-core/lib/errors.js:32:15)
Jun  3 12:59:58 raspberrypi CBatmo[4223]:     at Request.plumbing.callback (/home/pi/cbatmo/node_modules/request-promise-core/lib/plumbing.js:104:33)
Jun  3 12:59:58 raspberrypi CBatmo[4223]:     at Request.RP$callback [as _callback] (/home/pi/cbatmo/node_modules/request-promise-core/lib/plumbing.js:46:31)
Jun  3 12:59:58 raspberrypi CBatmo[4223]:     at Request.self.callback (/home/pi/cbatmo/node_modules/request/request.js:185:22)
Jun  3 12:59:58 raspberrypi CBatmo[4223]:     at Request.emit (events.js:198:13)
Jun  3 12:59:58 raspberrypi CBatmo[4223]:     at Request.<anonymous> (/home/pi/cbatmo/node_modules/request/request.js:1154:10)
Jun  3 12:59:58 raspberrypi CBatmo[4223]:     at Request.emit (events.js:198:13)
Jun  3 12:59:58 raspberrypi CBatmo[4223]:     at IncomingMessage.<anonymous> (/home/pi/cbatmo/node_modules/request/request.js:1076:12)
Jun  3 12:59:58 raspberrypi CBatmo[4223]:     at Object.onceWrapper (events.js:286:20)
Jun  3 12:59:58 raspberrypi CBatmo[4223]:     at IncomingMessage.emit (events.js:203:15)
Jun  3 12:59:58 raspberrypi CBatmo[4223]:     at endReadableNT (_stream_readable.js:1145:12)
Jun  3 12:59:58 raspberrypi CBatmo[4223]:     at process._tickCallback (internal/process/next_tick.js:63:19)
Jun  3 12:59:58 raspberrypi CBatmo[4223]:   name: 'StatusCodeError',
Jun  3 12:59:58 raspberrypi CBatmo[4223]:   statusCode: 400,
Jun  3 12:59:58 raspberrypi CBatmo[4223]:   message: '400 - {"error":"invalid_client"}',
Jun  3 12:59:58 raspberrypi CBatmo[4223]:   error: { error: 'invalid_client' },

exoticatom, 2020-06-03:
I am checking another post with similar problem.

I have defined .env:
```
APP_SECRET=ZZZZZZ
# Get a Netatmo credential from dev area : https://dev.netatmo.com/
NETATMO_CLIENT_ID=YYYYYY (email to netatmo site).
NETATMO_CLIENT_SECRET=XXXXXXX (my password to access netatmo site.)

```
It doesnt work.... cannot login

I see a post:
  body:
         `'client_id=YYYYYY&client_secret=XXXXX&grant_type=password&scope=read_station&username=mihajloz%40bluewin.ch&password=ZZZZZZ',`

but get:
```

        path: '/oauth2/token',
        _json: true,
        httpModule: [Object],
        agentClass: [Function],
        agent: [Agent],
        _started: true,
        href: 'https://api.netatmo.com/oauth2/token',
        req: [ClientRequest],
        ntick: true,
        response: [Circular],
        originalHost: 'api.netatmo.com',
        originalHostHeaderName: 'host',
        responseContent: [Circular],
        _destdata: true,
        _ended: true,
        _callbackCalled: true },
     toJSON: [Function: responseToJSON],
     caseless: Caseless { dict: [Object] },
     body: { error: 'invalid_client' } } }

```

exoticatom, 2020-06-03:
Oh. I made it work!

For everyone if wasn't clear as it wasnt to me:

When you log in to site then you would use:
`username  (email) and password to access netatmo site.`

You need to go to https://dev.netatmo.com/
and generate application, thats how you get:
```
- NETATMO_CLIENT_ID
-NETATMO_CLIENT_SECRET
```
plust:
`be sure that anything for APP_SECRET !!`

this goes to .env

am

Gulivertx, 2020-06-03:
Correct. I will update the readme according to your experience and be more clear about the netatmo dev center and what to do..

closed 2020-06-03 by Gulivertx