Gulivertx/cbatmo #30: Integration Status-Symbol of Sensors

karba71, 2020-06-02:
Hello Cedric,

i have 1 wish for your next version.
Can you integrate this status-symbols of the sensors?
I can't see this symbols on my mobile, only on my PC.

[image]

in this way:
[image]

Gulivertx, 2020-06-03:
Yes I can, would be available for next release.

Gulivertx, 2020-07-01:
Sorry but I did not have enough time these last days to release this with the version 2.3.2. I’m very busy for another huge project which must be done last month 😓

karba71, 2020-07-01:
Hello Cedric,

no prob, i can wait.

I will test your new version 2.3.2 in this time

Karba

Gulivertx, 2020-07-03:
ok I think I have something OK, not perfect but working
[image]

What do you think ?

karba71, 2020-08-06:
Hello Cedric,

looks like perfect!

Karba

Gulivertx, 2020-10-16:
Hi there, sorry for the long delay....
So I plane a new release for one of the next week which will integrate this.

closed 2020-10-23 by Gulivertx

karba71, 2020-11-20:
**Hello**

I have testet this days on my 7"-LCD (before only at my pc).
I think the status-symbols should be colored (similar netatmo pc-app).
Actually it's very difficult to distinguish.

[image]

[image]

Karba