Gulivertx/cbatmo #31: changing data in the graph

karba71, 2020-06-03:
Hello Cedric,

I can changing the data in graph by klick on the data:
I can take data from the 2. indoor-module (temp, CO2, acoustic), the the outdoor-module (temp, humidity, pressure), the rain-sensor and the wind-sensor.

But, if i take any data from the main-module cbatmo breakes.

Gulivertx, 2020-06-03:
I can reproduce this as well, will be fix with next release.

karba71, 2020-06-30:
Contrary to mine first post:

I can tap on the pressure, **it comes from the main-module** and works fine.
But the others (temp, co2, humidity) didnt work.

Is the problem not by the sensor rather by the gui?

Gulivertx, 2020-06-30:
it should work without any problem, did you use the branch deploy_script for your test ?

PR #40 (2020-07-01): New version 2.3.2
Fixes #31 #38 #35 #33 #32 #26
closed 2020-07-01 by Gulivertx