Gulivertx/cbatmo #32: graph didn't store selectet time

karba71, 2020-06-03:
Hello Cedric,

The graph will store which data i select, but not the time (12 hour, 1 day or 1 month).
It starts every time with 12 hour.

Gulivertx, 2020-06-03:
Are you sure about that? Rpi is slow to render month data. I ask because It’s ok on my side.

karba71, 2020-06-03:
I have check this.

I select 1 day or 1 month and i see all the data in the graph.
Then i do a reboot or close and reopen chromium.
The time in the graph is now 12 hour.

Gulivertx, 2020-06-03:
This is not a bug but a feature 🤭 
I’m kidding, it is correct I do not persist the last selected graph and scale, but it’s not a huge work to implement it. 

From my side it was not needed as my Raspberry stay turn on all the time with the app open. I only reboot for OS and cbatmo update...

karba71, 2020-06-03:
Hello Cedric,

I power the PI over a main switch from my AV-Receiver.
So i have a reboot once or twice a day.

To implement this would be nice.

karba

Gulivertx, 2020-06-05:
Ok I'm gonna implement a persist data system but only for UI user select. Will be available in a next release.

closed 2020-07-01 by Gulivertx