Gulivertx/cbatmo #35: TypeError after login

gfavre, 2020-06-07:
Hi,
and many thanks for this project. I've installed it both on my mac and raspberry. I get the same behavior on both machines after entering the credentials: black screen with a map and nothing else.
The browser's console displays "TypeError: undefined is not an object (evaluating 'e.replace')". Howser I can check on the network panel of the dev tools that it correctly get the token, and even gets the "netatmo station data" ([netatmo-station-data.json.zip](https://github.com/Gulivertx/cbatmo/files/4742091/netatmo-station-data.json.zip))

Same behavior on chrome, firefox or safari, using dev or prod env.

Do I need to setup anything else?

Gulivertx, 2020-06-07:
Hi, ho really strange, never had this problem before.
Did you clone the master branch ?

Could you just give me what you did to run cbatmo ?

If you run the app in DEV mode, replace .env **prod** by **dev** and run `yarn run watch` on you Mac can you give me which script and where the error is coming. Google Chrome developer tools shoud mention exactly where the error is when running the app id DEV mode.

gfavre, 2020-06-07:
Many thanks for your answer!

```
$ git log
commit ff514fdc5fb35aee7f4f13c3f810224734e45ae5 (HEAD -> master, origin/master, origin/HEAD)
```
=> not the commit / branch :(

```
$ yarn --version
1.22.4
```
=> seems pretty legit

```
$ yarn run build
...
lot of stuff happening, every line has [emitted] or [built]
```
here is the console:

[image]

Gulivertx, 2020-06-07:
Ok thanks. The crash append because the method removeAccent is call to an undefined value which should be a string. This value is your outdoor module name which seems not define because in your json response this value doesn’t exist. 
Could you try to specify a name of each of your modules from the official Netatmo mobile app and try again. Module names are mandatory to run cbatmo.

gfavre, 2020-06-08:
Many thanks for the support. Works like a charm!

Gulivertx, 2020-06-08:
Nice!
I will a do a change to allow undefined module names to avoid app crash for the next release. 

Thanks for your report.

closed 2020-07-01 by Gulivertx