Gulivertx/cbatmo #36: 2.3.1 Scrollbar flickering after x-hours

tdostler, 2020-06-18:
Hello,
first of all, big thanks for the great work. Now to the problem.

I use:
- Raspberry Touch-Screen as recommended (Optimized for a official Raspberry Pi touch monitor 7" (800x480))
- Chromium Browser
- After i opened the link (localhost), i switch with F11 to full-screen
--> All working well.

BUT, after x-hours (after the night), a scrollbar appears and flickering.
- Close Full-Screen (with F11) and reopen Full-Screen (with F11) does not help to solve the problem.
- I have to close the browser and open the localhost-site again.

What could be the problem? How can i solve it?
The screenshot shows the correct Layout. (Currently i have no screenshot when scrollbar appears after x-hours).  If you need, please contact me.
[image]

tdostler, 2020-06-18:
I saw, there is already a Issue #26 explaining my problem. It is planned to fix in 2.3.2. Therefore i close my Ticket. Thank you very much.

closed 2020-06-18 by tdostler

Gulivertx, 2020-06-18:
Yes problem is already known, I’m very busy these days to do any update sorry, on my way but still need more time to have some free time in order to fix it.