Gulivertx/cbatmo #39: new problem with yarn install --production

karba71, 2020-06-28:
Hello Cedric,

today i will test your fixes.
After transfer the zip i've got this output:

pi@PiDisplay:~/cbatmo $ yarn install --production
yarn install v1.22.4
[1/4] Resolving packages...
[2/4] Fetching packages...
error An unexpected error occurred: "/home/pi/.cache/yarn/v6/npm-core-js-3.6.5-7395dc273af37fb2e50e9bd3d9fe841285231d1a-integrity/node_modules/core-js/.yarn-metadata.json: Unexpected end of JSON input".
info If you think this is a bug, please open a bug report with the information provided in "/home/pi/cbatmo/yarn-error.log".
info Visit https://yarnpkg.com/en/docs/cli/install for documentation about this command.

Whats the problem?

Gulivertx, 2020-06-30:
hmm it is not related to cbatmo, or seems not.
could you try t oclean yarn cache and run again ?

`yarn cache clean`
then
`yarn install --production`

karba71, 2020-06-30:
Hi,

 

„yarn cache clean“ works - thanks

closed 2020-06-30 by Gulivertx