Gulivertx/cbatmo #45: V2.3.4; Station-Name overlays Time

karba71, 2020-11-19:
Hello Cedric,

Thanks for update to V2.3.4 with integrate of this status-symbols.

In this version the name of the station is very long and overlays the Time from the last data.
I think this was better in V2.3.3 (and i know my adress).

V2.3.3
[image]

V2.3.4
[image]

Karba

Gulivertx, 2020-11-20:
I did not change anything on this part in the last version. Station Uhlestrasse 34.... is your station name, you should rename your station in the Netatmo Application.
And what is coming inside the brackets is the main module name but it is not a choice from my side but a change that Netatmo did on their API, the station name now concatenate this value between bracket. Unfortunately this not something I have the hands on.
I could trim the name but this is not something I want to do myself. I dod not know why Netatmo change that!

karba71, 2020-11-20:
Hello Cedric,

 

thanks.

 

I have check this.

 

V2.3.3 shows this long name too.

I think Netatmo have change here anything.

 

 

Karba

karba71, 2020-11-20:
Hello,

fixed, it was my mistake!

V2.3.3 its the same - i have change the name and then (the name is to long) we have this overlay.

Karba

karba71, 2020-11-20:
fixed, it was my mistake!

closed 2020-11-20 by karba71