Gulivertx/cbatmo #46: status-symbols should be colored

karba71, 2020-12-05:
**Hello**

I think the status-symbols should be colored (similar netatmo pc-app).
Actually it's very difficult to distinguish.

Status-Symbole 20 11 20
[image]

Karba

Gulivertx, 2020-12-05:
Hi,
Sounds like good idea, where did you find the graphic resources?

karba71, 2020-12-05:
Hello,

this pictures are from my netatmo pc-app.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

You mean this symbols?
Some symbols are missing, i will look for this.

karba71, 2020-12-05:
[image]

karba71, 2020-12-06:
[image]

karba71, 2020-12-06:
[image]

karba71, 2020-12-06:
[image]

[image]

karba71, 2020-12-12:
[image]
[image]
[image]
[image]

karba71, 2020-12-12:
Hello,

I think this are all symbols.

Karba

Gulivertx, 2020-12-15:
hmm, I think I cannot use this graphic resources, as they are propriety of Netatmo. I will look after the official developer documents if there is some information about. Anyway, thanks for the image.

closed 2020-12-28 by Gulivertx