Gulivertx/cbatmo #50: Scrollbar in version 2.3.4

okhoel, 2020-12-15:
Tried installing the latest version (2.3.4) downloaded today (2020-12-15), and I get a scrollbar on the official Raspberry Pi Display:
[image]

I have a feeling that this might have to do with the fact that I don't have a second indoor module, so the graph and the wind are on separate "lines".

Bodokan, 2020-12-15:
Hi okhoel

I had the same issue.
My solution was to start the chromium with the following paramter:

sudo nano /etc/xdg/autostart/chromium.desktop  

[Desktop Entry]
Type=Application
Name=Chromium
Comment=Chromium Webbrowser
NoDisplay=false
Exec=chromium-browser --noerrdialogs --disable-crash-reporter --disable-infobars **--force-device-scale-factor=0.94** --kiosk "http://192.168.x.x:3000"

okhoel, 2020-12-15:
> Exec=chromium-browser --noerrdialogs --disable-crash-reporter --disable-infobars **--force-device-scale-factor=0.94** --kiosk "http://192.168.x.x:3000"

Thanks for the tip.
Tested adding the scale-factor to my command, and it works fine as a workaround.
I suppose it would be good to have a permanent fix anyway.

Bodokan, 2020-12-15:
of course this is only a workaround ...

Gulivertx, 2020-12-15:
hi there,

Hum this problem seems to come from wind module which is normally places at the bottom of indoor module. And in your case it takes the place of the Rain module.
I think the problem come from default height of Wind module which is bigger than the Rain, to have a completely compatible layout the two lines should have exactly the same height.

I will investigate if I can do something quickly or If I need more effort to fix this.
Thank's for your report.

Gulivertx, 2020-12-15:
by the way I use webkit-gtk in place of chromium it could also come from their... Could you try in webkit-gtk on your side if you also have the same issue ?

okhoel, 2020-12-15:
I could try webkit-gtk... But I have never used it, and I can't figure out how to install it.
Maybe if you point me in the right direction...

Levanterman, 2020-12-15:
I had a similar issue, for a quick/temporary fix I reduce the font size of the sub-label element by 0.1 from 0.9 to 0.8 in the style sheet, file  style.scss which shrinks it enough for the scroll bar to disappear (rebuilt/deploy)

[image]

Gulivertx, 2020-12-28:
> I could try webkit-gtk... But I have never used it, and I can't figure out how to install it.
> Maybe if you point me in the right direction...

Hmm I do not remember by heart the debian packages dependencies needed and do not have any access to my installation (yeah sky holidays :) ). When I'm back to home I will give you the instruction to install and use it.
There is already a script in the repo to use it but you need first install the dependencies.
https://github.com/Gulivertx/cbatmo/blob/master/deploy/webkit/browser

Gulivertx, 2021-01-08:
> Hmm I do not remember by heart the debian packages dependencies needed and do not have any access to my installation (yeah sky holidays :) ). When I'm back to home I will give you the instruction to [...]

Ouch sorry forget about that. Here is how you can use webkit-gtk

```bash
sudo apt install gjs gnome-themes-standard libwebkit2gtk-4.0-37 libwebkit2gtk-4.0-37-gtk2 libwebkit2gtk-4.0-dev
```

Then id the deploy folder of cbatmo you have a script to run it, just use like this:
```bash
cd cbatmo/deploy/webkit/
./browser --fullscreen http://localhost:3000
```

okhoel, 2021-01-11:
Then I have tried webkit-gtk. The issue is the same in the way that some of the text goes below the screen. The difference is that the scrollbar is only visible when the mouse pointer is all the way to the right.
[image]
[image]

Gulivertx, 2021-01-12:
Ok thanks for your test, so I think I know from where the problem is coming. I will try to fix it and do an update during the next week.

closed 2023-04-27 by Gulivertx