Gulivertx/cbatmo #52: Manual Build fails

exoticatom, 2020-12-23:
Hi, can you pls help.
I did manual build on my PC and transferred ZIP to raspi.
When I execute "yarn install --production" it fails:
[image]

any idea what could it be ?

thanks

exoticatom, 2020-12-23:
Maybe this could help ?
[image]

exoticatom, 2020-12-23:
oh... lesson learned 
`sudo aptget -install yarn `

doesn't installs the correct version..

use: [Link to installation instructions](https://classic.yarnpkg.com/en/docs/install/#debian-stable)

Maybe update documentation? Yarn doesnt come preinstalled on raspi.

Gulivertx, 2020-12-28:
Correct and you already had this problem in the past : https://github.com/Gulivertx/cbatmo/issues/29 ;)

exoticatom, 2020-12-28:
oops , I clicked closed :-) Not sure if it should be like that.

yes, and I completely forgot about it. I had to rebuild the new version. Strange that I am the only one always having this issue with a strange yarn version.

Gulivertx, 2021-01-05:
information about yarn added in README

closed 2021-01-05 by Gulivertx