Gulivertx/cbatmo #54: missing translation

karba71, 2020-12-29:
Hello,

I noticed today a missing translation.
My PC shows "Regensensor" for my rain-gauge.
cbatmo (V2.3.4)  shows "Rain".

[image]

Karba

uwewiest, 2020-12-30:
Already fixed with 2.3.5 like it seems

[image]

Gulivertx, 2020-12-31:
This should be because you do not give any name for your module. When the module name is empty I set a default name in English only (https://github.com/Gulivertx/cbatmo/blob/master/src/models/NetatmoNAModule3.ts#L43). 
If in this case you want the default name in German could you please give me the all modules names in German then I will add them in the translator files ?

Here is the all module :
- Main
- Outdoor
- Wind
- Rain
- Indoor

uwewiest, 2020-12-31:
Well here you go:

    Main = Haupt
    Outdoor = Aussen
    Wind = Wind
    Rain = Regen
    Indoor = Innen

uwewiest, 2020-12-31:
Ah yes in you src de file I did see you have used in your common.json file für German (de)
 I can only suggest to maybe use the wording like here:

{
  "notifications": {
    "loading": "Startet...",
    "configuration_success": "Konfiguration abgeschlossen, bitte warten...",
    "configuration_error": "Oops, es ist ein Fehler aufgetreten!",
    "not_reachable": "Dieses Modul kann die Station nicht erreichen. Ueberpruefen Sie den Batteriestatus oder bringen Sie es naeher an die Station."
  },
  "app_info": {
    "description": "Netatmo Wetterstation fuer Raspberry Pi mit offiziellem Touchscreen.",
    "version": "Version",
    "created_by": "Erstellt von"
  },
  "forecast" : {
    "forecast": "Vorhersage",
    "sunrise": "Sonnenaufgang",
    "sunset": "Sonnenuntergang"
  },
  "netatmo": {
    "barometer": "Barometer",
    "time": "Zeit",
    "graph": "Diagramm",
    "day": "Tag",
    "month": "Monat",
    "temperature": "Temperatur",
    "humidity": "Luftfeuchtigkeit",
    "co2": "CO2",
    "pressure": "Druck",
    "cumulative": "Kumulativ",
    "rain": "Kumulativ",
    "noise": "Schallpegel",
    "wind_strength": "Windgeschwindigkeit",
    "windstrength": "Windgeschwindigkeit",
    "wind_max_day": "Max Tag"
  }
}

karba71, 2020-12-31:
Hello,

but I have a name for every sensor:

[image]

Karba

uwewiest, 2020-12-31:
Yes me too and the are in the view above in the screenshot ...or what do you mean?

karba71, 2020-12-31:
Hello Uwe,

the left one are the names from the Netatmo-App from my Win-PC.
There is a name for any sensor.
The right one are the names from the cbatmo-App.
cbatmo uses the correct names except the rain-sensor.
Here comes the default-name.

[image]

uwewiest, 2020-12-31:
Hmm I have 2.3.5 and it says Regenmesser for me

Auf Deutsch:

Schau oben auf meinem Screenshot...der Name stimmt bei mir Regenmesser

Gruß

Uwe

karba71, 2021-01-01:
Hallo Uwe,

zuerst noch ein gesundes neues Jahr 2021 - auch an alle anderen!!!

Ich habe noch die V2.3.4.

Kann sein das das in der 2.3.5 gefixt wurde, die Änderungsbeschreibung
"Minor fixes
Create new types folder to centralize types for typescript
Bump Webpack to 5.x + fix webpack config
Bump React to 17.x
Upgrade all other NPM packages to their last version"
lässt es nicht vermuten. Deshalb habe ich die auch gar nicht getestet.

Karba

uwewiest, 2021-01-01:
Ja das denke ich auch ich habe gestern Mal alles auf den neuesten Stand gebracht selbst bpm selbst hatte eins

Also wie ganz oben geschrieben ich denke auch 2.3.5 hat das schon gehoben

Gruß

Uwe

Gulivertx, 2021-01-12:
So what is the point for this issue ?
I'm not facing to any issue to get my module name. FYI I use of course the last version 2.3.6

Please if you're not up to date do an update of your installation and give me a feedback and we will start to debug together (by Teamviewer)

uwewiest, 2021-01-12:
I don't think there is one Cedric all good :-) works fine

karba71, 2021-01-12:
**Hello,**

i have read in your changelog for V2.35: 
-Bump Webpack to 5.x + fix webpack config
-Bump React to 17.x
-Upgrade all other NPM packages to their last version
I havent read:
-Minor fixes

My mistake – Sorry for this.

Because of this non functional changes i have report this missing translation.
Uwe have report that this point is fixed in V2.35.

But:

I have an old picture from 20.11.2020; V2.34.
Here i can see „Regenmesser“:

[image]

And with the same V2.34 i have today "Rain":

[image]

Netatmo shows "my" names:

[image]

I think there WAS a issue at V2.34.

I will update to V2.36 the next days and check this.

Karba

Gulivertx, 2021-02-03:
@karba71 : any feedback ? Can I close this issue ?

closed 2021-02-21 by Gulivertx