Gulivertx/cbatmo #55: difference betweeen Netatmo and cbatmo at WiFi-Status

karba71, 2020-12-29:
Hello,

cbatmo; V2.3.4

There is a difference betweeen Netatmo and cbatmo at WiFi-Status from my main-station (Wohnzimmer).
I think I have a good WiFi-Signal and the WiFi-Status is Full.
But cbatmo shows a lower signal.

[image]

[image]

Karba

Gulivertx, 2020-12-31:
Hi, thanks to your report.
Hmm interesting, I actually used this information : https://dev.netatmo.com/apidocumentation/weather
Look at section Specifications.
Then in my code I used the given values : https://github.com/Gulivertx/cbatmo/blob/master/src/models/NetatmoNAMain.ts#L101-L114

Gulivertx, 2021-01-04:
Should now be fixed with version 2.3.6

closed 2021-01-04 by Gulivertx