Gulivertx/cbatmo #56: Select different station ?

jorisbertomeu, 2020-12-30:
Hey, I would to know if it was possible to choose an other weather station ?
I have 3 weather stations within my Netatmo account but an unwanted station is shown within your interface.

Maybe can I change this option somewhere within the code ?

Merci 😄

Gulivertx, 2020-12-31:
Hi,
Currently there is no way from the application to select a specific station. I never had fellows using multi-stations.
I can implement the station selector, I will create a new git branch for multi-stations and will need you to test if the change will work.

I will update this post when a beta version will be ready for test.

TBR-BRD, 2021-01-10:
Hi, i have the same problem. I am eagerly awaiting the solution.

TBR-BRD, 2021-01-21:
Please help

Gulivertx, 2021-02-03:
> Please help

Hi, sorry but currently didn't have anytime to spend on cbatmo :'(
I know that you wait about this and I will do it as soon as I would be able to do. The sooner will be better.
If someone want to help and propose a modification I will also be happy :p

Gulivertx, 2021-02-12:
Hi, I started the new branch [dev_multi_station](https://github.com/Gulivertx/cbatmo/tree/dev_multi_stations).
The modification is not so easy and need a complete rework of data structure to handle correctly multi-stations.
I could make something one the current structure but it would not be nice and complicated for the maintenance of the code.
Currently this branch is on the beginning of the change and not works ! Do not try it, I will telle you when I would need tries from your side as I don't have two stations.

Gulivertx, 2021-02-21:
hi guys,
In the branch dev_multi_station there is a version which allow to change the station when more than one station is link to an account.
Could you please make a try of this branch and give me a feedback.
As I said I cannot test this change as I do not have multiple stations.

If you need any help to try it just ask my for help and we will look together. Thanks in advance for your test!

closed 2022-02-17 by Gulivertx