Gulivertx/cbatmo #80: Cbatmo is runnonut stage, but browser stuck in loading

jukkaforss, 2021-09-22:
This started suddenly, it has working perfectly for months. 

After raspberry pi reboot, browser stuck in loading status. 
cbatmo service is running. 

pi@raspberrypi:~ $ systemctl status cbatmo.service
* cbatmo.service - CBatmo WEB server
   Loaded: loaded (/etc/systemd/system/cbatmo.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2021-09-22 11:49:52 UTC; 41min ago
 Main PID: 369 (node)
    Tasks: 23 (limit: 2059)
   CGroup: /system.slice/cbatmo.service
           |-369 node /usr/share/yarn/bin/yarn.js start
           |-924 /bin/sh -c node server.js
           `-925 /usr/bin/node server.js

Sep 22 11:49:52 raspberrypi systemd[1]: Started CBatmo WEB server.
Sep 22 11:50:01 raspberrypi CBatmo[369]: yarn run v1.22.5
Sep 22 11:50:02 raspberrypi CBatmo[369]: $ node server.js
Sep 22 11:50:13 raspberrypi CBatmo[369]: Server running on http://localhost:3000 as production
pi@raspberrypi:~ $

Any idea what could be wrong?

jukkaforss, 2021-09-22:
[cbatmo.pdf](https://github.com/Gulivertx/cbatmo/files/7210640/cbatmo.pdf)

Gulivertx, 2021-09-24:
Hi,

- Try to clean browser cache
- From a another computer reach CBATMO from Google Chrome or another web-browser and see if you also have the same issue, if yes open the dev tools and look after any Javascript error to post here

jukkaforss, 2021-09-24:
`i18next: languageChanged en
bundle.437619a9224327bb3054.js:2 i18next: initialized Object
ContentScript.js:112 Exception in onResRdy: TypeError: Cannot read properties of undefined (reading 'htmlRes')
192.168.0.59/:1 Unchecked runtime.lastError: Could not establish connection. Receiving end does not exist.
5Failed to decode downloaded font: <URL>
192.168.0.59/:1 Failed to decode downloaded font: http://192.168.0.59:3000/fonts/c422543605bc94796ef2c68787f08d05.woff
192.168.0.59/:1 Failed to decode downloaded font: http://192.168.0.59:3000/fonts/c422543605bc94796ef2c68787f08d05.woff
192.168.0.59/:1 Failed to decode downloaded font: http://192.168.0.59:3000/fonts/c422543605bc94796ef2c68787f08d05.woff
192.168.0.59/:1 Failed to decode downloaded font: http://192.168.0.59:3000/fonts/c422543605bc94796ef2c68787f08d05.woff
192.168.0.59/:1 Failed to decode downloaded font: http://192.168.0.59:3000/fonts/c422543605bc94796ef2c68787f08d05.woff
5OTS parsing error: invalid sfntVersion: -2147399670
192.168.0.59/:1 OTS parsing error: invalid sfntVersion: -2147399670
192.168.0.59/:1 OTS parsing error: invalid sfntVersion: -2147399670
192.168.0.59/:1 OTS parsing error: invalid sfntVersion: -2147399670
192.168.0.59/:1 OTS parsing error: invalid sfntVersion: -2147399670
192.168.0.59/:1 OTS parsing error: invalid sfntVersion: -2147399670
bundle.437619a9224327bb3054.js:2 Fetch station data
bundle.437619a9224327bb3054.js:2 Uncaught (in promise) TypeError: t.json is not a function
    at bundle.437619a9224327bb3054.js:2`

jukkaforss, 2021-09-24:
I tried it already with external PC and browser and result was same.
Messages from dev console are above.

jukkaforss, 2021-09-24:
[error_message.txt](https://github.com/Gulivertx/cbatmo/files/7225559/error_message.txt)

If I click last message, it shows this.

jukkaforss, 2021-09-24:
Logged in using VNC and started Chromium once. did go to google.com and that worked fine.
After this cbatmo loads fine.

Not sure what changed?

closed 2021-09-25 by jukkaforss