Gulivertx/cbatmo #91: Netatmo API Security Update

tdostler, 2022-08-11:
Hi Cedric,

Netatmo will change the API!

Extract from the mail i got:
Dear Netatmo developer,
To improve the security of our products, we inform you that the Client Credentials grant type method will be completely removed. It will no longer be possible to authenticate with the username and password of the user.The effective date of this update is October 2022.How can you authenticate with Netatmo API ?From this date, the OAuth2 authorization code flow must be followed for authentication.You can find details on this method on our website:

I assume that from October onwards, this will not working anymore? Do you will update accordingly in advance?

BR
Thomas

Gulivertx, 2022-08-12:
Hi,
Yep I also received this information.
Currently I did not have any time to move to oauth2 authentication.
What I can underrstand, this is only for the authentication process.
If you are already logged in with password grant (the current used on CBatmo), you should be ok to still continue to use the app, this if your refresh token is not in its end of life. It's should be ok because the refresh token still get an update of its end of time each time it is use to refresh the access token.

For sure, to used CBatmo for a new installation, an update it's needed :(

If you have time and knowledge to do it I would really appreciate to have a pull request with the change ;)

I also not used anymore this app, I created a new Dashboard app with Netatmo, Philips hue, me heating system and my solar pv system, then for me, I have to move this project first.

tdostler, 2022-08-12:
Hello Cedric,

 

i hope you are right, and it still works after October 2022. We will see.

I´m not able to change your coding to change to oauth2, sorry for that.

 

Regarding the "new Dashboard" you described: This sounds very good.

Do you have it also in GitHub?

 

BR

Thomas

 
 

Gesendet: Freitag, 12. August 2022 um 12:19 Uhr

Gulivertx, 2022-08-12:
It's on GitHub, but on a private repository. Currently I did not plane to share this project because the code is not really clean. Everything works fine, but it's more about the code :p

it's how is looking...

[image]
[image]

RRyyas, 2022-12-26:
Is that also why i wasn't able to log in on the app at all? I've initiated my NetAtmo API key earlier today, and nothing happened when trying to log in to the app

Gulivertx, 2022-12-26:
> Is that also why i wasn't able to log in on the app at all? I've initiated my NetAtmo API key earlier today, and nothing happened when trying to log in to the app

Unfortunately yes this is why.

closed 2023-04-27 by Gulivertx